Ultimaker/Cura #18450: Inconsistent selection of extruder

captain118, 2024-02-27:
### Cura Version

5.6.0

### Operating System

Windows 1

### Printer

Snapmaker 2.0

### Reproduction steps

1. Assign model to print using extruder 2
2. Modify settings on extruder 1
3. Print

### Actual results

It prints using extruder 1

### Expected results

I should print using extruder 2

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/14413657/cura.log)

GregValiant, 2024-02-27:
Thanks for the report.
Please use the "File | Save Project" command and zip the resultant 3mf project file and post the zip file here.  Without knowing your settings this cannot be investigated.

closed 2024-03-12 by github-actions[bot]

captain118, 2024-03-12:
Sorry for the delay I hadnt been able to do any test prints lately.
[S2ADE_gridfinity-rebuilt-baseplate-ikea-drawer-1.zip](https://github.com/Ultimaker/Cura/files/14574236/S2ADE_gridfinity-rebuilt-baseplate-ikea-drawer-1.zip)

captain118, 2024-03-12:
[image]
From the image you can clearly see that Extruder 1 is disabled.

GregValiant, 2024-03-12:
Thanks for the file.  I don't have a SnapMaker definition file so only the model loads.  There have been some recent reports about this so I'll relate them.
When you disable Extruder 1 the "Support" and "Skirt Brim" extruder options become hidden.  The bug is that Cura will use those settings and if they are "not overridden" or set to "Extruder 1" then Cura will use those even when E1 is disabled.  That can mean that a print will start with T0 even though all it does is one lap of a skirt.

You can try this workaround.
Re-enable E1.
Explicitly set the "Support" and "Skirt/Brim" extruders to E2.
Disable E1.

That should do it.  If it does not then paste your StartUp G-Code here.  Sometimes explicitly setting the Tools in the StartUp can lead to odd results in the gcode.

The Cura team is aware of this bug and hopefully it will be fixed in 5.7.0.