Ultimaker/Cura #18455: Print head collision

SR2205, 2024-02-27:
### Cura Version

5.5

### Operating System

Windows 10

### Printer

Sovol sv04

### Reproduction steps

Due to the experimental filament I am using I had to add a 20mm fan on he0. When he1 is activated it moves all the way to the left, but it now has 10mm less space to move. This is an unnecessary movement written into the system prior to actually printing with he1. I need to disable this movement to stop the collision. This is not a bug but there is no other venue to request support on this issue. 

### Actual results

bang crash belt grind

### Expected results

 no bang crash belt grind

### Add your .zip and screenshots here ⬇️

[jpeg.zip](https://github.com/Ultimaker/Cura/files/14426275/jpeg.zip)

Asterchades, 2024-02-28:
I've adjusted the flags on this post to indicate it isn't actually a bug, but I cannot offer anything more than speculation in regards to a solution. IDEX isn't something I've got any experience with, unfortunately.

That said, I can't find anything within the included configuration files for the SV04 which might alter this behaviour. At no time does it define the size of the tool head, nor does it appear to have any specific code for changing which tool is in use. As such my suspicion is that the movement that's causing you grief is determined by the firmware of your printer itself, putting it outside of the ability for Cura to actually do anything about what you're encountering.

A quick check sort-of confirms this, as the Marlin `Configuration_adv.h` file does define an X2_MIN_POS setting which includes the comment "A min coordinate so the X2 carriage can't hit the parked X1 carriage" that sounds exactly like what's happening here. There does not appear to be any means to change this value via commands, which means you'll likely need to compile and upload a new firmware build in order to correct this behaviour.

_Note that if you've changed from the default Marlin firmware to an alternative, such as Klipper, it should still have such facilities. It's just likely to be easier to reconfigure on-the-fly than Marlin._

SR2205, 2024-02-28:
That is a really well stated answer, thanks! I've been working through so
much "customer support" that is basically useless that I'm actually taken
aback here now that I got a really good answer. I'll keep digging for
solutions, in the mean time, thanks again!

you think I could solve this with klipper? I've never used it but I'm
willing to learn it if thats the case

Asterchades, 2024-02-28:
From what I can see it should be addressable in Klipper as well, yes. It would also be easier to fine tune, given it offers the ability to change the entire configuration without creating a new firmware build (Marlin's significantly more modest hardware requirements mean many less-changed settings embedded when compiling). However it also appears to require more user configuration as it has no "default" behaviour for what happens when a tool is enabled or parked - user-created macros are used instead when changing between tools.

Unfortunately it appears that neither Klipper nor Marlin provide default configuration files for the SV04. I'm almost certain you could find something from a different source but there's nothing on the official repositories beyond a generic IDEX example and other, single-extruder Sovol printers.

Again, though, I feel the need to emphasise that IDEX (or multiple extruders in general) is outside of my knowledge base, and I lack the appropriate hardware to validate anything relating to its firmware implementation.

GregValiant, 2024-02-29:
My knowledge of IDEX printers is right there with Asterchades.  Ultimaker doesn't make an IDEX printer so there isn't any native support for them in Cura.
I've played around with "virtual" IDEX machines in Cura and have gotten some to work.  A lot does depend on the firmware and now you have a "Custom" unsupported printer.

What is this gizmo here?  Does it happen to be an endstop switch for the right head?
[image]

If it is the endstop switch then maybe you can shim it out so the right head hits the switch before it crashes?

There are a couple of things you can try.  If the printer supports A and B axes (or alternately V and W) then you might be able to adjust the homing to account for the new distance between the nozzles.
G28 X Y followed by G28 A B (or V W) might produce a different result.  This would be dependent on the endstop switch for HE1.
If it's possible to adjust the Home Offsets separately for each print head you might be able to keep them from crashing during a print.  It wouldn't help with homing though.

Would it be possible to design a new bracket for the fan and move it up and in?  It would require a custom duct as well.  That would be my first choice because changing the firmware is always a slippery slope.
This is obviously my printhead and it would have the same problem.  Moving the fan up and in would fix it.
[image]