Ultimaker/Cura #18456: Anycubic Kobra Pro Fails most slices

Ellevehc86, 2024-02-28:
### Cura Version

5.6.0

### Operating System

Windows 10 

### Printer

Anycubic Kobra Pro

### Name abnormal settings

None 

### Describe model location

Middle of baseplate 

### Describe your model

3D terrain print 

### Add your .zip here ⬇️

[WWA_SHNN.zip](https://github.com/Ultimaker/Cura/files/14428108/WWA_SHNN.zip)

Ellevehc86, 2024-02-28:
I try to print it uaing basic setting, "normal' fails during slicing. I didn't have the same issue with my Ender 3 and printed the same file

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic