Ultimaker/Cura #18457: [5.6.0] unnecessary strings generated

plgarcia, 2024-02-28:
### Cura Version

5.6.0

### Operating System

Windows 10 Family 22H2

### Printer

FLSUN V400

### Reproduction steps

Cura generates unnecessary strings. Nothing fancy, grid support generated, that's all.
Load the project, slice and then preview.
It is not a display bug, the strings are really printed. 
[image]
 
[FV_T-RexSkull_JawBug.zip](https://github.com/Ultimaker/Cura/files/14433698/FV_T-RexSkull_JawBug.zip)
[cura.log.zip](https://github.com/Ultimaker/Cura/files/14433794/cura.log.zip)
Let me know if more information is needed.
;End of Gcode
;SETTING_3 {"global_quality": "[general]\\nversion = 4\\nname = Normal #2\\ndefi
;SETTING_3 nition = flsun_v400\\n\\n[metadata]\\ntype = quality_changes\\nqualit
;SETTING_3 y_type = normal\\nsetting_version = 22\\n\\n[values]\\nadaptive_layer
;SETTING_3 _height_variation_step = 0.04\\nadhesion_type = skirt\\narcwelder_all
;SETTING_3 ow_dynamic_precision = True\\narcwelder_allow_travel_arcs = True\\nar
;SETTING_3 cwelder_enable = True\\nbridge_settings_enabled = True\\nlayer_height
;SETTING_3  = 0.3\\nlayer_height_0 = 0.3\\nmaterial_linear_advance_enable = Fals
;SETTING_3 e\\nsupport_enable = True\\nsupport_structure = normal\\nsupport_type
;SETTING_3  = everywhere\\n\\n", "extruder_quality": ["[general]\\nversion = 4\\
;SETTING_3 nname = Normal #2\\ndefinition = flsun_v400\\n\\n[metadata]\\ntype = 
;SETTING_3 quality_changes\\nquality_type = normal\\nsetting_version = 22\\nposi
;SETTING_3 tion = 0\\n\\n[values]\\ninfill_enable_travel_optimization = True\\ni
;SETTING_3 nfill_pattern = gyroid\\nironing_enabled = True\\nmaterial_initial_pr
;SETTING_3 int_temperature = 210.0\\nmaterial_print_temperature = 210.0\\nskirt_
;SETTING_3 line_count = 3\\nsupport_pattern = grid\\nsupport_xy_distance = 0.16\
;SETTING_3 \n\\n"]}

### Actual results

Cura generates unnecessary strings.

### Expected results

No string

### Add your .zip and screenshots here ⬇️

[cura.log.zip](https://github.com/Ultimaker/Cura/files/14433988/cura.log.zip)

HellAholic, 2024-03-01:
Thanks for the report.
I can see the strings, added a ticket CURA-11685 to the backlog to investigate this.

Cheers!

GregValiant, 2024-03-01:
The Support Z Distance in the project is 0.20 and the Layer Height is 0.30.  Setting the "Support Z Distance" to an exact multiple of the Layer Height seems to fix this.

A workaround is setting the "Support Interface Wall Line Count" to a non-zero number.

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant