Ultimaker/Cura #18460: Moon lamp fails to slice

lpulley, 2024-02-28:
### Cura Version

5.7.0-alpha.0

### Operating System

Windows 10

### Printer

Anycubic Vyper

### Name abnormal settings

Fails with default profiles and custom profiles

### Describe model location

Slicing works when rotated 90 degrees

### Describe your model

The model has been used by lots of people but may not be watertight

### Add your .zip here ⬇️

https://we.tl/t-Cd294OhkSn

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

lpulley, 2024-03-01:
You mean more fixes than in the 5.7.0 alpha 2 days ago?

HellAholic, 2024-03-01:
[image]
Yes, there are few more fixes.