Ultimaker/Cura #18463: Error when splicing with Wellbot Magana se

ividal1978, 2024-02-29:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Helbot Magena SE

### Name abnormal settings

It bakes with default profile 0.2

### Describe model location

It brakes in any psiion

### Describe your model

Is aBudha seating

### Add your .zip here ⬇️

[HMSEPRO_BudhaMix2.zip](https://github.com/Ultimaker/Cura/files/14441685/HMSEPRO_BudhaMix2.zip)

GregValiant, 2024-02-29:
Thanks for the report.
I couldn't get the project to slice.  After repairing the model I still had to change the "Initial Layer Height" to 0.20 to get it to slice.
Error code is 3221225477
[cura5-7.zip](https://github.com/Ultimaker/Cura/files/14448397/cura5-7.zip)

The Cura team will take a look.

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic