Ultimaker/Cura #18464: Makerbot Method 1A and 2A extruder in Cura

CJA98, 2024-02-29:
### Is your feature request related to a problem?

I have a MakerBot Method printer and cannot use Cura 5.6.0 as a slicer due to lacking material and/or extruder settings.

### Describe the solution you'd like

It would be nice to be able to add the 1A and 2A extruders for PLA and PVA to either a new profile for the Method or to the Method X profile so that it's possible to use them. When I try slice and load the file there is a tool and or material mismatch and it is not possible to print.
I would like to be able to slice with Cura becasue of all the possibilities for setting compared to the cloud print slicer. 

### Describe alternatives you've considered

I've used the cloud printing software but it lacks the settings to further better my prints. I have tried slicing in Cura 5.6.0 with the wrong extruder and filament configured but these give me an error message both with USB on the printer and in the cloud print software.

### Affected users and/or printers

Only Makerbot Method

### Additional information & file uploads

[image]
[image]

GregValiant, 2024-02-29:
@nallath is this something that the Cura team is aware of?

CJA98, 2024-02-29:
I'm not sure. I also wasn't sure where I should post and or feedback this...

CJA98, 2024-02-29:
Thanks for the reply as well!

GregValiant, 2024-03-01:
_"Fan speed needs to be a 0-1 value otherwise the Method doesn't do anything with it"_

If the "Printer Settings" plugin is loaded (it's in the Marketplace) the bottom setting is "Scale Fan Speed To 0 - 1".