Ultimaker/Cura #18465: Typical G-Code modification

Stefavia, 2024-02-29:
### Is your feature request related to a problem?

No possibility to choose Klipper in g-code flavor

### Describe the solution you'd like

Add klipper g-code

### Describe alternatives you've considered

Add klipper g-code

### Affected users and/or printers

Anycubic Kobra 2 Max

### Additional information & file uploads

[image]

GregValiant, 2024-02-29:
This is a duplicate of #12171.  I have read some things here about making adjustments for Klipper.  Maybe @jellespijker can comment on that.

HellAholic, 2024-03-01:
We have a ticket but not sure when we get around to implementing it, for internal reference: 
CURA-10397

Cheers!