Ultimaker/Cura #18470: Multi material

Jarren-Essentium, 2024-02-29:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ultimaker S5

### Name abnormal settings

No changes, Default PLA 

### Describe model location

Center of bed fails and other rotations/locations/scales fail

### Describe your model

I've sliced this model many times before with earlier versions of Cura ~5.0 + without any slice errors. 

### Add your .zip here ⬇️

I've tried this model https://www.thingiverse.com/thing:1394269/files before with earlier versions. This fails on different printers and with other complex models like the 2 color bull https://www.thingiverse.com/thing:1442085. The failure happenes when only 1 toolhead is used and when both toolheads are used. Also it fails with and without support generated and with and without a purge tower.
[UMS5_2color_treefrog_small_hollow_fixed.zip](https://github.com/Ultimaker/Cura/files/14452718/UMS5_2color_treefrog_small_hollow_fixed.zip)

Jarren-Essentium, 2024-02-29:
I also tried the nightly build and still got a slicing error.

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic