Ultimaker/Cura #18473: Printer's print time is significantly faster compared to Cura's estimate

nick1213mc, 2024-02-29:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Creality Ender 3 V3 SE

### Reproduction steps

Imported a Model into Cura (with the preset of the "Creality Ender 3 V3 SE" at the default PLA and 0.20mm layer height) and sized it to my liking. I then spliced it and Cura gave the estimate time to print of about 8 Hours

### Actual results

The estimate time in Cura was significantly slower than the actual print time of my printer (the print itself only took about 4-5 Hours)

### Expected results

While having a fast printer is a good thing in most respects, I was trying to essentially print something to be as big as it could be within an 8 hour window. Had I known it would have been only 4-5 hours, I would have increased the size dramatically

I was just looking that Cura's estimate time would be at least fairly close to the overall final printing time of my printer, for project/printing planning

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-03-01:
Thanks for the report.
The "estimated time vs actual time" thing comes up a lot.  It is always a function of the settings and what the printer has in it's firmware which is something that Cura can't know.

We ask for a project file because it would contain your model, your printer, and all your settings.  Things like not enabling Acceleration in Cura can cause wild swings in the time estimate.  If the printer has low Max Accel then a print will take longer than the estimate.  If you don't enable Accel control then Cura really has to guess as the software has no way to tell what numbers the printer is going to use.  In that case the print can finish faster than the estimate.

My printer (Ender 3 Pro) consistently finishes within 2 or 3 percent of the Cura estimate.

nick1213mc, 2024-03-02:
Ahh sorry about that. I added a project file here, hopefully that helps.

According to the website, the acceleration of the printer is 2500mm/s2, typical printing speed is 180mm/s, and max printing speed is 250mm/s. 

https://www.creality3dofficial.com/products/ender-3-v3-se-3d-printer 

And my apologies if this is incredibly basic stuff; I'm very new to printing and didn't want to do something to Cura to make my printer slow down or otherwise mess with it's operation

[Turtle Duck.zip](https://github.com/Ultimaker/Cura/files/14470022/Turtle.Duck.zip)
.

GregValiant, 2024-03-02:
You need to use the "File | Save Project" command to create a project file.
It looks like you "exported" the model as a 3mf file.  All that 3mf has is the model.
With the standard profile, support enabled, and the model at 200% I'm showing a 14:54:00 print.  That's with the print speed at 180.

GregValiant, 2024-03-02:
I played with this a bit more.
With an E3 V3 SE active, support enabled, and the model sized at 200% - Cura shows the print time at almost 15 hours.  That's with the print speed at 180mm/sec.
Switching to one of my Ender 3 Pro profiles and the print speed at 75mm/sec I'm getting a Cura estimate of 5:12.  If I increase the speed of my profile to 180 the print time drops to 4:27
So it looks like Creality did something in that definition file for the V3 SE that Cura doesn't like or can't account for.

At any rate, this doesn't appear to be a Cura problem but rather a problem created by the author of the definition files.  The definitions for non-Ultimaker printers come from a member of the community who owns one, or from the manufacturer.  All the Creality definitions were originally created by the author "Trouch.com".

I'll remove the bug label and close this.  I hope you understand.

closed 2024-03-02 by GregValiant