Ultimaker/Cura #18474: Slicing failed with an unexpected error Cura 5.7 Alpha

Handydoug, 2024-03-01:
### Cura Version

Full Version Cura 5.7 Alpha

### Operating System

Linux Ubuntu 22.04.4 LTS

### Printer

Creality Ender 5 Plus

### Name abnormal settings

No and No

### Describe model location

No and No

### Describe your model

Yes, sliced successfully before. Don't know if watertight. 

### Add your .zip here ⬇️

[CE5P_2-28short handle threads.zip](https://github.com/Ultimaker/Cura/files/14457970/CE5P_2-28short.handle.threads.zip)

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic

Handydoug, 2024-03-02:
Hello,
I can't slice a simple cube.
What should I do until this gets fixed?
Thank you,
Doug