Ultimaker/Cura #18480: Strut Print

jjhartz, 2024-03-01:
### Cura Version

5.6 

### Operating System

Windows 11

### Printer

Ultimaker 3

### Name abnormal settings

No

### Describe model location

I tried all possible orientations, print heads etc.

### Describe your model

A simple straight strut, shaped like a piece of angle iron

### Add your .zip here ⬇️

[UM3_Strut-A.zip](https://github.com/Ultimaker/Cura/files/14465468/UM3_Strut-A.zip)

GregValiant, 2024-03-01:
Thanks for the report.
There have been a lot of problems reported with slicing in Windows 11.  I'll mark this as a duplicate.
I didn't have any trouble slicing it with 5.6.0 and Windows 10.

jjhartz, 2024-03-01:
Wow, that is disappointing. What is the recommendation for this Windows 11
problem? Revert to Windows 10 or is there some kind of Cura fix in
the works?

Thanks,
John

GregValiant, 2024-03-01:
The problem sort of appeared out of nowhere.  I don't know if it will be fixed for 5.7.  I do know the Cura team has been working on it.
I don't know if the problem occurred with previous versions.  If you go [HERE ](https://github.com/Ultimaker/Cura/releases?page=2) you can download an earlier release.  I thought 5.2.2 was pretty good.  Scroll down to it and click on the "Assets" link below the blurb.