Ultimaker/Cura #18493: simple box does not slice

holgerlembke, 2024-03-02:
### Cura Version

5.7.0-alpha

### Operating System

windows 10

### Printer

creality ender 3

### Describe your model

prusa slicer does it fine.

### Add your .zip here ⬇️

[kiste.zip](https://github.com/Ultimaker/Cura/files/14470180/kiste.zip)

cura 5.7.0-alpha-0

GregValiant, 2024-03-04:
Thanks for the report.
I can't duplicate the problem.  The project sliced fine for me.  This is on a Dell laptop running Win10Pro and Cura 5.6.0.

This is not the first report regarding "Can't slice anything".  There have been some files that simply won't slice for me either.  I don't know if this is because of some compatibility problem between Windows and Cura or something inside Cura itself.

The Cura team will take a look.
Here is the gcode generated from your project.
[kiste.zip](https://github.com/Ultimaker/Cura/files/14485061/kiste.zip)

holgerlembke, 2024-03-04:
yepp, 5.6.0 slices it perfectly.

but read my report: Cura Version 5.7.0-alpha

GregValiant, 2024-03-04:
5.7alpha is a nightly build for testing only.  It isn't an "official release" and it is going to have issues.
You can leave a comment [HERE](https://github.com/Ultimaker/Cura/discussions/18080) if you like.  It's the discussion page for problems with the nightly build.

holgerlembke, 2024-03-04:
Greg, this is all soooo confusing.

5.6.0. sometimes crashes, https://github.com/Ultimaker/Cura/discussions/18080 says I shall use the nightly 5.7.0.alpha and when the nightly crashes a window says I shall report the error here and the button takes me to this place to open an issue...

If this issue is wrong here, someone should close it with a link to https://github.com/Ultimaker/Cura/discussions/18080.

GregValiant, 2024-03-04:
Yes it is confusing.
I guess that the Cura team had some fixes they wanted to test.  People running Windows 11 really seem to be having problems with Cura.  I suppose the idea was to test some of the fixes.  Putting the alpha out gives them a broader base of testers, but it is confusing when it comes time to make a report.
If there is an issue with 5.7 I'd report it on the discussion page with a quick description and a project file, and if it's 5.6.0 it can go here in the issues page.