Ultimaker/Cura #18494: Bug in initial_extruder_nr function when using Brim and Skirt

x40-Community, 2024-03-02:
### Cura Version

5.6.0

### Operating System

Linux

### Printer

Weedo X40

### Reproduction steps

The model should be printed with an index printer from the right extruder. To set the extrider, the following is used in the start script:
T{initial_extruder_nr} S ; Set start extruder

### Actual results

With Build Plate Adhesion Type "None" and "Raft" the correct result with T1 S is generated in gcode.
However, when using "Brim" and "Skirt" the result is T0 S and therefore the wrong extruder!

### Expected results

Regardless of the Build Plate Adhesion Type, the correct extruder should come out at the initial_extruder_nr function.

### Add your .zip and screenshots here ⬇️

[Archiv.zip](https://github.com/Ultimaker/Cura/files/14470195/Archiv.zip)

GregValiant, 2024-03-03:
Thanks for the report.
The bug I see here is that the Skirt/Brim extruder _must be explicitly set_ to Extruder 2 when that is what you want.
Whether Extruder 1 is enabled, or disabled, "initial_extruder_nr" will return a "0" when the Skirt/Brim extruder is either "Not Overridden" or "Extruder 1".
With Extruder 1 disabled and the skirt/brim extruder explicitly set to Extruder 2 I get:
T1
M82 ;absolute extrusion mode
;--------------------------StartUp Gcode
**The Initial Extruder is: #1**

With Extruder 1 disabled and the skirt/brim extruder "not overridden":
;Generated with Cura_SteamEngine 5.6.0
T1
M82 ;absolute extrusion mode
;--------------------------StartUp Gcode
**The Initial Extruder is: #0**

@casperlamboo has looked at this previously (see bug report #17501) and it does appear that the behavior is a bit different.  One of the problems I see is that if Extruder 1 is disabled, the Skirt/Brim extruder settings are hidden.  The user has to understand that the settings will be used even when hidden.  It is necessary to re-enable Extruder 0, set the skirt/brim extruder to Extruder 2, and then disable Extruder 1 again.

x40-Community, 2024-03-03:
I forgot something important. The left extruder (T0) is deactivated! Overriding the brim and skirt is therefore not adjustable. Despite deactivated T0, T0 is written into the gcode as an extruder for brim and skirt.

If both extruders are active and the right extruder has the brim or skirt creation set to the right extruder, then it works.

HellAholic, 2024-03-08:
@x40-Community 
I'm checking on the latest internal build and I cannot reproduce the issue using another printer (bibo dual) which uses the same syntax for `M109 T{initial_extruder_nr} S{material_print_temperature_layer_0, initial_extruder_nr}`
when I disable the first extruder (T0) it switches correctly to T1 as the initial extruder with skirt or brim.
[image]
[image]

As the Weedo X40 is not part of the Cura repo I cannot confirm that it fixes your issue as well.

closed 2024-03-08 by HellAholic