Ultimaker/Cura #18496: Update the project name when launched with a new STL in single instance mode.

lttlrck, 2024-03-02:
### Is your feature request related to a problem?

When using 3D Print in Fusion 360 to directly open an STL in Cura **in single instance mode** the name is not updated. 

[image]

F3D creates a file based on the object name.

First print is fine, it takes the name of the file:

[image]

Subsequent prints do not update the name:

[image]

Why is this a problem?? Because of Klipper :-)

I print directly to Klipper:

[image]

Cura uses the project name for the job name, this means Klipper is showing the "wrong" name, but also Klipper history is effectively broken because the names are reused and Klipper _it does not retain the STL for consecutive duplicate names_  

[image]

The workaround is of course to rename before printing:

[image]

Which is honestly fair enough, but it'd be _really_ nice is Cura could just update the name, it's the only glitch in this very nice workflow.

### Describe the solution you'd like

I'd like it to update the name when launched with a new file in single instance mode.

### Describe alternatives you've considered

The workaround above.

### Affected users and/or printers

Anyone using F3D 3D Print to Cura.

GregValiant, 2024-03-03:
Thanks for the suggestion.
This sounds like a duplicate of bug report #11276 (and others).  Take a look at that one and let us know if it describes what you are asking for.

It appears that this has not been fixed and I also find it very annoying.  It is much too easy to inadvertently over-write a gcode file.

I'll leave this open as a feature request but I'll add a bug label and the Cura team will take a look.

lttlrck, 2024-03-03:
> Thanks for the suggestion. This sounds like a duplicate of bug report #11276 (and others). 

It's not the same but I can imagine it's very closely related so good to mark it as such.