Ultimaker/Cura #18497: Add variable nozzle sizes to Creality Ender 3 S1 Pro

Ki6FHX, 2024-03-02:
### Is your feature request related to a problem?

Cura doesn't provide any option other than .4 mm nozzle to choose from. 
Therefore I have to try to guess at which attributes to change
Cura is the best, but I can understand why some may find this too daunting of a task and want to change slicer once they purchase different size nozzles and want to print with those.

### Describe the solution you'd like

Please add drop down of nozzle sizes for Creality Ender 3 S1 Pro (and I think any Ender 3 printer from what I understand).
I know other printers have  these other sizes available.
.1 mm
.2 mm
.4 mm
.6 mm
.8 mm
1.0 mm

### Describe alternatives you've considered

I think adding the current list of nozzle sizes is the best because I think there are Cura settings connected to these nozzle sizes.
However, if it's not that easy, then provide documentation on how to make changes based on a nozzle size that isn't supported within Cura

### Affected users and/or printers

I think it affects any Creality Ender 3 printer.

Asterchades, 2024-03-03:
You can always adjust the nozzle size of a printer via the `Extruder` tab of the `Machine Settings` under the `Manage Printers` dialogue. It's a little bit buried as it's not a setting that's normally changed with any great frequency, but it's still there and available to be set to whatever you please.

For quicker access you could also use the `Printer Settings` plugin, which adds numerous settings related to your printer to the settings list on the right when in advanced/custom mode.

For a feature like this, however, it needs to be provided by community submissions. _All_ third party definitions - including the addition of variants, such as nozzle sizes - are handled via pull requests as the Cura team simply lacks the time, resources, and appropriate hardware to maintain the definitions for anything outside of the Ultimaker (and now MakerBot, following the merger) printers. Each printer in this case needs to have the variants separately added, and no inheritance is (currently) possible.

With that said, the appropriate variants for the S1 Pro _are_ present in the configuration files, and have been since the S1 Pro was added. I can only begin to guess as to why they're not showing up in your local installation as you've lodged this as a feature request rather than a bug report, so understandably not included a log file.

GregValiant, 2024-03-03:
All of the Creality definitions include nozzle sizes.
As Asterchades mentions, they are here for the S1 Pro:
[image]

I'll go ahead and close this.  If you don't have the nozzle sizes as indicated then you might need to re-install Cura.  The files are there and should be accessible.  They are in the "C:\Program Files\UltiMaker Cura 5.6.0\share\cura\resources\variants\creality" folder.

closed 2024-03-03 by GregValiant