Ultimaker/Cura #18500: slice failed when moved down -1 z axis

moviebrain, 2024-03-03:
### Cura Version

5.6.0

### Operating System

windows 10

### Printer

wanhao i3

### Name abnormal settings

.4 nozzle with 1.0mm width

### Describe model location

center

### Describe your model

https://www.thingiverse.com/thing:4733825/remixes

### Add your .zip here ⬇️

sliced as-is, but when i tried to move the z-axis down 1mm to prevent one side from generating support it failed
[Pegboard Cordless Tool Wall Mount - 4733825.zip](https://github.com/Ultimaker/Cura/files/14472055/Pegboard.Cordless.Tool.Wall.Mount.-.4733825.zip)

GregValiant, 2024-03-03:
Thanks for the report.
I am unable to get the model to "not slice".
Set Cura up to slice and use the "File | Save Project" command, zip the resultant 3mf file, and post the zip folder here.

BTW a 0.4 nozzle has a 1.0 diameter flat on the end.  You can push the line width to 1.0, but the material squish will be limited to the diameter of the flat spot and that can cause a very rough print.

moviebrain, 2024-03-04:
[Desktop.zip](https://github.com/Ultimaker/Cura/files/14476860/Desktop.zip)

sure. here is the 3mf as requested. project fails, and a screen shot of my machine settings that shows the nozzle is NOT "blank"

GregValiant, 2024-03-04:
Thanks for the files.
It still slices for me with the part at Z=-1 (and everywhere else).  The model doesn't seem to have any errors.
Have you tried moving it on the build plate and/or rotating it?

The Cura team will take a look.
[image]

moviebrain, 2024-03-04:
I sure appreciate the replies. I ended up getting it to slice by leaving it at 0,0,0 and then multiplied it to two models. It was honestly the very first time cura flat out refused to slice something and thought I’d make the big report. It happened on both 5.6 and the 5.7 alpha I was prompted to try.

closed 2024-03-04 by moviebrain