Ultimaker/Cura #18505: Version 5.6.0

MBE23, 2024-03-03:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

ZONESTAR Z9V5

### Reproduction steps

Après découpe, le bouton n'est pas traduit (Save - enregistré)

### Actual results

rien

### Expected results

rien

### Add your .zip and screenshots here ⬇️

Pas de traduction voir image (SaveTo Disk  -> Enregistré sur le disque)
[image]

Vandresc, 2024-03-08:
Hi @MBE23 , we fixed that a while ago, the fix will be included in the next release, thank you for your feedback
[image]

closed 2024-03-08 by Vandresc

MBE23, 2024-03-26:
Thanks, this is fixed in version 5.7.0-beta.1