Ultimaker/Cura #18508: print has extra print layer that sticks out of the obj

TrueBigBomb, 2024-03-04:
### Cura Version

5.2.1

### Operating System

windows 11 version 22H2 (OS build 22621.3155)

### Printer

creality ender-3 / ender-3 v2

### Reproduction steps

i even updated to the latest version of cura and loaded in my project and it didnt fix it

### Actual results

the project/obj has layers that shouldn't be there. i relaced it with a new duplicate and that fixed it.

### Expected results

fixed it somehow

### Add your .zip and screenshots here ⬇️

[CE3E3V2_stake.zip](https://github.com/Ultimaker/Cura/files/14486699/CE3E3V2_stake.zip)
[image]
[image]
[image]
[image]
[image]

GregValiant, 2024-03-05:
Thanks for the report.
This is a weird one.  There is something wrong with that model.  I know it's the same as the other piece with the same name, but look what happens when I move the "good" model a little bit:
[image]

I exported the model and opened it in MS 3D Builder.  An error was found in the model so I had the app repair it.  When I brought the repaired model back into Cura that "bad" spot was gone.

This sure looks like a model problem rather than a Cura bug.  I'm inclined to remove the bug label and close this.  Let me know if you feel that is premature.
The model on the far right has been repaired and doesn't show the problem.
[image]

TrueBigBomb, 2024-03-06:
> Thanks for the report. This is a weird one. There is something wrong with that model. I know it's the same as the other piece with the same name, but look what happens when I move the "good" model a [...]

i did ask a group i am in and they said the same thing. so would this be a error/ bug that fusion 360 needs to fix.

closed 2024-03-06 by TrueBigBomb

GregValiant, 2024-03-06:
Sometimes this happens when a model is put together.  Merging pieces into a single model can also cause the problem if there is even a small gap between the pieces being merged.   The pieces need to be properly constrained to keep that from happening.
"Loft" and "Shell" operations generate complex surfaces and the translation to STL can be a problem.  I don't know if I can consider that a bug.  It is up to the CAD designer to create models that are properly designed.

Usually the problem occurs at an interface between two features.  In this case the model shows a problem in the middle of nowhere.  I'm not sure why that occurred.  Something could have been altered and the surface wasn't properly closed?  

It is always prudent to check the exported models with a repair utility (Fusion has NetFabb which is pretty good) to insure that the model is free of errors before you start spending time on slicing.  You assumed that the model was good and the error was in the slicer.  My experience doing the triage here is that 90% of the time it is the other way around.
That being said, there does appear to be a "won't slice" bug in Cura when run on Win11 systems.  The symptom is that a slice will not complete.  Since that wasn't the problem here I immediately suspected it was a model problem.