Ultimaker/Cura #18511: Impossible to insert a support blocker in this model

MarcelloMazza, 2024-03-05:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

FLSUN V400

### Reproduction steps

Good morning,
I don't know why but in this project it is practically impossible to insert a support block to avoid printing the support inside the empty cylinder.
The procedure I follow is: support in all possible points > I insert the support block and size it up to what is necessary to avoid printing of the support where it should be excluded > sectioning

### Actual results

The support is also printed inside the hollow cylinder where I put the support block

### Expected results

There should be no support printed in the hollow cylinder
This is the first time I haven't been able to insert the support block into a model

### Add your .zip and screenshots here ⬇️

[CAlciatore balilla blu anteriore.zip](https://github.com/Ultimaker/Cura/files/14492181/CAlciatore.balilla.blu.anteriore.zip)
[Errore supporto - Calciatore balilla.zip](https://github.com/Ultimaker/Cura/files/14492182/Errore.supporto.-.Calciatore.balilla.zip)

GregValiant, 2024-03-05:
Thanks for the report.
The setting for "Support Horizontal Expansion" is set to 0.8 so the support is wider than the supported area.  Because of that, the support for the "shoulders" is growing from the base.
[image]

If you set "Support Horizontal Expansion" to zero the condition should go away.
[image]

Let us know if that works for you.  I'll remove the bug label as this is expected behavior and a setting issue.

MarcelloMazza, 2024-03-05:
Perfect: problem solved. Many, many thanks

closed 2024-03-05 by MarcelloMazza