Ultimaker/Cura #18515: Check for automake busted.

zBeeble42, 2024-03-05:
### Cura Version

main branch

### Operating System

FreeBSD-CURRENT

### Printer

Custom

### Reproduction steps

install conan, configur conan (as instructed, but with CC=clang16), clone git repository, 

Run conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv

... last line of output:

ERROR: libtool/2.4.7 require 'automake/<host_version>': didn't find a matching host dependency

Then I check for automake:

[2:92:164]dgilbert@strike:~/devel/Cura> pkg which `which automake`
/usr/local/bin/automake was installed by package automake-1.16.5_1

further check:

[2:93:165]dgilbert@strike:~/devel/Cura> automake --version
automake (GNU automake) 1.16.5

### Actual results

The system can't be built.

### Expected results

The system can be built.

### Add your .zip and screenshots here ⬇️

[image]

jellespijker, 2024-03-08:
It seems that you're running into an issue with a Conan package provided by the Conan-Center-Index (CCI) You could try to report an issue there https://github.com/conan-io/conan-center-index/issues?q=is%3Aissue+is%3Aopen+

Also make sure that all system dependencies as described in the Wiki are installed.
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source#linux 

Let me know it this fixed your issue

closed 2024-03-22 by github-actions[bot]

ErikTempelaarVO, 2024-08-14:
I believe this has been solved in conan, see here:
https://github.com/conan-io/conan/issues/14704