Ultimaker/Cura #18519: Failed slicing in both Cura 5.6 Release & 5.7 Alpha

Rowenther, 2024-03-06:
### Cura Version

5.6.0

### Operating System

Windows 11 Home

### Printer

Ender 3 S-1 Pro

### Name abnormal settings

I have used these settings to slice a 8x8 version of this model

### Describe model location

Attempted to rotate, 90 & 45 w/o success

### Describe your model

Model analysis shows its watertight

### Add your .zip here ⬇️

I have successfully sliced an 8x8 version of this model & printed successfully. File attached is a reduced triangle version of the original I tried to slice & failed as well. This version has approx 1/4 less triangles.

[2x8 Multiboard Side Tile - top-bottom peg - Reduced 2.zip](https://github.com/Ultimaker/Cura/files/14508727/2x8.Multiboard.Side.Tile.-.top-bottom.peg.-.Reduced.2.zip)

GregValiant, 2024-03-06:
Thanks for the report.
That model looks really bad in the Prepare screen of Cura.  If I may assume that those are threads in the square feature and that the walls of the hexes should be chamfered and flat - this would not print correctly even if it did slice.
[image]

Speaking of slicing, I didn't have a problem on my Win10 laptop.  There have been recurring problems on Win 11 systems.
[image]

What software did you use to reduce the triangles?  MS 3D Builder has a "Simplify" function and I don't recall it doing anything like that to my models.

LilBub, 2024-03-07:
I'm having the same crashes, see [https://github.com/Ultimaker/Cura/issues/18541]