Ultimaker/Cura #18522: Error when changing only the layer height value

Berhert76, 2024-03-06:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Biqu B1

### Name abnormal settings

Alterado apenas a altura de camada de 0.45 para 0.30

### Describe model location

Usei o addon de teste de overhang. Não foi mexido a posição do modelo na mesa.

### Describe your model

Apenas um modelo de teste de overhang inserido do addon PARTS OF CALIBRATION.

### Add your .zip here ⬇️

O fatiamento simplesmente não acontece com uma altura de camada de 0.3 para bico 0.6.
Se eu mudo para 0.45 a altura de camada, ele funciona.
-
[image]
-
**Download de arquivos** >> [Cura_slicer_problem.zip](https://github.com/Ultimaker/Cura/files/14510794/Cura_slicer_problem.zip)

GregValiant, 2024-03-06:
Thanks for the report.
I didn't have any trouble slicing it at 0.3.  There have been a number of reports lately regarding problems when tree supports are enabled.  Try turning off "Generate Support" and see if the problem goes away.  That particular model has not been a problem.

Berhert76, 2024-03-07:
Tks Greg!