Ultimaker/Cura #18526: Have a per profile Start and End Gcode

gbarbehenn, 2024-03-06:
### Is your feature request related to a problem?

Printing a large, dimensionally accurate print job is very different from printing a small verification print.  The first may take several hours and so a G29/M420 isn't inconvenient.  The latter may take less than an hour to print, so G29/M420 is a significant time waster.

### Describe the solution you'd like

Ideally each profile would have its own Start/End Gcode.  Since this would be a hassle to configure for all profiles, maybe the existing Start/End Gcode could be the default, only overridden if the profile has its own set of Start or End Gcodes?

### Describe alternatives you've considered

The only alternative is use one set of Start/End codes, or edit the Start/End codes per job.

### Affected users and/or printers

Everybody is affected by this as they have to choose a less than optimal Start/End codes, or remember to edit them per job.

GregValiant, 2024-03-07:
This is doable in Post-Process and I've written a script that disables bed leveling if the print is below a certain minimum footprint on the bed.  It doesn't take time or height into account.
I'll take another look.  Adding a "minimum time" shouldn't be a problem.

GregValiant, 2024-03-07:
"Little Utilities" is a collection of 14 seldom used but handy post-processors that people have requested here and there. 
Unzip the file and place "LittleUtilities_GV.py" into the "scripts" folder in your Configuration Folder (use "Help | Show Configuration Folder" to find it).  It will be available under "Extensions / Post Process / Modify Gcode" and then "Add a Script".
The one you will want is "Disable ABL for small models".  Enable it and then enable "Minimum print time" and enter your break point in minutes.  Prints that are shorter will have the G29 and/or M420 commented out of the startup gcode.  A message will come up when you save the gcode telling you whether or not the ABL will run.
[LittleUtilities_GV.zip](https://github.com/Ultimaker/Cura/files/14526552/LittleUtilities_GV.zip)
I don't know if this will ever make it into Cura.  Putting it into the "scripts" folder should ensure that updating Cura doesn't over-write it.

GregValiant, 2024-03-23:
Can this be closed?

closed 2024-04-06 by github-actions[bot]