Ultimaker/Cura #18530: Slicing - Unexpected error

gregreenwood, 2024-03-06:
### Cura Version

5.6.0 & 5.7.0-alpha.0

### Operating System

Linux Mint 21 Cinnamon

### Printer

Longer LK5 Pro

### Describe model location

No

### Describe your model

Not in Cura.  It slices OK in Prusa Version 2.7.1+linux-x64-GTK3

### Add your .zip here ⬇️

[SDI_Power_Shell-Body.3mf.zip](https://github.com/Ultimaker/Cura/files/14516919/SDI_Power_Shell-Body.3mf.zip)

GregValiant, 2024-03-07:
Thanks for the report.
On Windows 10 it was one of the fastest slices I've seen in a long time.  The model is free from errors.

I'll mark this as a duplicate and the Cura team will take a look.
[image]

gregreenwood, 2024-03-08:
I was using LMint V21 Cinnamon and the applications were appimages.

gregreenwood, 2024-03-09:
I fixed the issue by deleting the /home/[my user
name]/.local/share/cura/5.6 folder and redownloaded the appimage for 5.6
and ran it.  Now it slices.

Regards
GG

gregreenwood, 2024-03-09:
Correction.  The bug returned when I loaded CuraEngine Tiled Infill
Generation 0.2.0 from the Marketplace!
I started with Honeycomb and it failed to slice.  Choosing anything after
that, it fails no matter what I choose.
Uninstalling the Tiled Generation 0.2.0 fixes the failure to slice.

Hope find the bug.
Regards
GG

GregValiant, 2024-03-10:
That will help.  Thanks for adding it.