Ultimaker/Cura #18532: Cura 5.6 fails to finish slicing a really detailed/large/tall model

DuckersMcQuack, 2024-03-07:
### Cura Version

5.6

### Operating System

Windows 10 home

### Printer

custom ender 3 pro with skr 1.3 running klipper

### Reproduction steps

Loaded up cura, set maximum deviation to 0.005, same with minimum polygon circumference, load a articulated dragon that has 2.55 million triangles, and slice it. cura 4 will crash because it can't process more than less than 8GB ram, or it crashes, or cura 5.6 i tested, it will get stuck half way and stay there in a bad loop.

### Actual results

Cura 4.12.1 crashes at 8GB ram usage (i prefer cura 4.12 because none of the addons i use has been updated for 5 yet, and 5 has an annoying speed limit/smooth scroll i don't like)

Cura 5.6 gets stuck in a processing loop and ends up dropping in ram usage little by little and never gets past 50% sliced. 

### Expected results

Expected it to slice the model.

I can not upload the project because it contains a paid model. I have included the profile used though, and  I can show a screenshot of it. 
[image]

### Add your .zip and screenshots here ⬇️

[duckers ender 0.15 jan 24.zip](https://github.com/Ultimaker/Cura/files/14518436/duckers.ender.0.15.jan.24.zip)
[image]

GregValiant, 2024-03-07:
Thanks for the report.
One of the worst models to ever show up here was called a "Crystal Dragon".  It had thousands of errors in it.  It was repairable in MS 3D Builder (you can download it for free from the Microsoft store).  After it was repaired it would slice.
Your model reminds me a lot of that one.  Try opening it in 3D Builder and see if it needs repair.
This could well be a slicing bug but without being able to check the model myself, you will need to do it.
Also, try turning off supports completely just as a test.
Let us know if repair fixes the problem or if "no supports" frees up the slice.

DuckersMcQuack, 2024-03-07:
> Thanks for the report. One of the worst models to ever show up here was called a "Crystal Dragon". It had thousands of errors in it. It was repairable in MS 3D Builder (you can download it for free [...]

Aye, i remember those, but thankfully i always throw my models into 3d builder to check for errors before slicing them. 

cura 4.12 does not report model errors, but 5.6 does. 

Sliced both now at 150mm height as that's needed to get a full 3 lines solid first smallest joint, but cura 4 crashed, and 5.6 gets stuck in a loop, shoots up to 40GB, then slowly drops down to 7GB, then increased to 30GB and stays there.

Also, what does the "highlighted areas" look like? As i've seen this warning many times, but can't seem to find those. (those "errors" are after 3dbuilder has repaired it)

[image]

[image]

GregValiant, 2024-03-07:
The "highlighted areas" are sometimes so miniscule that they don't show up in the view.  Typically larger ones will be a "polka-dot" pattern on the model like on the one below.  Cura might ask you to look at the model in "X-ray" view.  That works if the errors are gross (they will show up as red surfaces) but not so well when they are tiny.
This is a horse of a different color and it would not slice without repair.
[image]