Ultimaker/Cura #18535: I do not see anything

Lexenlewax, 2024-03-07:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Anycubic Kobra and other

### Reproduction steps

Hello Guys, 
If I load an STL file I do not see anything. I do not see my print bed or the object. I added also another Printer. Still I do not see anything. I can slice and print. But I cannot change the angle or see the slices. Please check the attachment. The Object is sliced. But I do not see anything. 
[image]

Thank you for your support!

### Actual results

I can print. But I do not see anything. 

### Expected results

I want to see the object. I want to move it on the print bed. 

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-03-07:
Thanks for the report.
I haven't heard of this before.
In the bottom left of the Cura workspace are 5 icons to adjust the "view".  Click on the middle one and see if the "top down" view brings the bed and model into view.
Maybe something is going on between Cura and your video system of that computer.
Tell us more about the computer.  Does it have integrated graphics, a dual system, a high end video card (like NVidia)?
Does it have an integrated mouse or is it a USB mouse?

Lexenlewax, 2024-03-07:
Hello, this is a standard laptop with an integrated graphic dot I used the laptop and US emails was scrolling wheel. I clicked on every button. I still do not see anything. Sorry for the shot and sass but I am driving in the car.Von meinem iPhone gesendet

GregValiant, 2024-03-08:
Did Cura work previously on that computer?
You can try going to "Preferences / Configure Cura" and in "Viewport Behavior" enable "Force Layer View Compatibility Mode" and restart Cura.
I think there is a miscommunication between Cura and your video system.  There are a lot of Cura users and this is the first time this problem come up.

Lexenlewax, 2024-03-08:
Hello Greg, 

I am a beginner. This is the first time that I installed Cura on my computer. 

 

I changed the setup according to your suggestion. Unfortunately it did not help. 

 

I even do not see the printer bed. 

 

I installed Version 5.5.0 and it works now. 

 

Thank you for your support!

 

Best Regards, 
Alexander Rifel

Vandresc, 2024-03-08:
Hi @Lexenlewax , could you provide us with the logs? Maybe we can find something in there that may cause this issue. As Greg mentioned, we never heard of this issue before. 
You can find your log file here:
Windows: %APPDATA%\cura\<Cura version>\cura.log
or while in Cura you can navigate to Help -> Show configuration folder

Lexenlewax, 2024-03-11:
For this I would need to reinstall 5.6.0. right?

Vandresc, 2024-03-14:
If you removed all old config for 5.6 then I am afraid that you will need to reinstall it yes