Ultimaker/Cura #18538: Cura loads but Window doesn't show on Linux

unluckynelson, 2024-03-07:
### Cura Version

5.6.0

### Operating System

Garuda Linux

### Printer

Creality 3 pro

### Reproduction steps

Run the appimage

Terminal output:
```
λ ./UltiMaker-Cura-5.6.0-linux-X64_cbc4dfea87e29f7635e925d1ba0769c0.AppImage 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
QApplication: invalid style override 'kvantum' passed, ignoring it.
	Available styles: Windows, Fusion
QQmlApplicationEngine failed to load component
QQmlComponent: Component is not ready

** (UltiMaker-Cura:8904): WARNING **: 17:30:03.264: atk-bridge: get_device_events_reply: unknown signature
```
It seems like it is running but with no window

Here's the log files:
[cura.log](https://github.com/Ultimaker/Cura/files/14526592/cura.log)

Tried installing:
sudo pacman -S qt6-wayland

### Actual results

The splash screen shows and finishes loading but the window doesn't open

### Expected results

I expect the main window to display

### Add your .zip and screenshots here ⬇️

[image]

unluckynelson, 2024-03-07:
The suggestion in this issue solved my problem: https://github.com/Ultimaker/Cura/issues/12940

Running the appimage with `QT_STYLE_OVERRIDE=""` worked for me

Vandresc, 2024-03-08:
Hi @unluckynelson , glad the workaround did work for you.

closed 2024-03-08 by Vandresc

doobes, 2024-03-08:
> The suggestion in this issue solved my problem: #12940
> 
> Running the appimage with `QT_STYLE_OVERRIDE=""` worked for me

Thanks.

How, exactly, is this done?

I'm a bit of a Linux neophyte

chris

unluckynelson, 2024-03-08:
You just run it inline with the appimage.$> QT_STYLE_OVERRIDE="" ultimaker_cura.appimageSent from my Galaxy

doobes, 2024-03-09:
Nope.

[image]

doobes, 2024-03-09:
Nope:

[image]

unluckynelson, 2024-03-09:
Before calling appimage. Not afterSent from my Galaxy

Screenshot.from.2024-03-09.10.33.45.jpeg (view on web)