Ultimaker/Cura #18541: 5.7 alpha crashes slicing print

LilBub, 2024-03-07:
### Cura Version

5.7a

### Operating System

Windows 11

### Printer

CR-10S (custom)

### Name abnormal settings

Tree Supports

### Describe model location

Tilted back 45 degrees (used Auto Orientation Plug In)

### Describe your model

Yes, previously fine in 5.6 and 5.7a

### Add your .zip here ⬇️

Only real changes have been to optimize tree support settings.  I had changed the Support Z distance from 0.4 to 0.2   Now Cura crashes everytime on on 5.6 and 5.7a

[CCR10S_helmet1 proj 57a.zip](https://github.com/Ultimaker/Cura/files/14528075/CCR10S_helmet1.proj.57a.zip)

GregValiant, 2024-03-08:
Thanks for the report.
Cura has not been playing nice with Windows 11.  I'll mark this as a duplicate and the Cura team will take a look.
This is in 5.7alpha but on a Win10 computer.

LilBub, 2024-03-08:
> Thanks for the report. Cura has not been playing nice with Windows 11. I'll mark this as a duplicate and the Cura team will take a look. This is in 5.7alpha but on a Win10 computer. !

I managed to get this to slice on another Windows 11 computer that's a bit more powerful (Intel i9 , 64gb RAM, RTX 3070, etc)  It did take a few tries.

Of note, ,when installing Cura 5.7a on that PC, I could not get past the 'add a machine' process.  I noticed with Cura launched, i didnt see the usual 'Loading machines' step.  I uninstalled and reinstalled, then it worked.  One in a million?