Ultimaker/Cura #18547: Slicing Failed

alnavasa, 2024-03-08:
### Cura Version

5.6.0

### Operating System

Mac OS 13.6.3 (22G436)

### Printer

S5 R1

### Name abnormal settings

Standard settings

### Describe model location

lay flat 

### Describe your model

Car Aircon grill

### Add your .zip here ⬇️

[UMS5_BMW Grille  2.3mf.zip](https://github.com/Ultimaker/Cura/files/14537073/UMS5_BMW.Grille.2.3mf.zip)
[image]

GregValiant, 2024-03-08:
Thanks for the report.
I couldn't get this to slice in either 5.6.0 or 5.7alpha.
Cura 5.2.2 and PrusaSlicer didn't have any trouble with it.

I'll mark this as a duplicate and the Cura team will take a look.

alnavasa, 2024-03-08:
can I downgrade to 5.2.2 to print on ABS on Ultimaker s5?

GregValiant, 2024-03-08:
You should be able to.  I have found 5.2.2 to be the best of the 5.x versions.

alnavasa, 2024-03-11:
working perfectly with 5.2.2

closed 2026-09-11 by alnavasa