Ultimaker/Cura #18548: initial flow rate

Paolinter, 2024-03-08:
### Cura Version

5.6.0

### Operating System

windows11

### Printer

elegoo neptune 4

### Reproduction steps

i tried to set a different flow for the initial layer, i set it to 110%.

### Actual results

When I save the file, there is no reference to the  differente flow. I looked for inside the G code the M221 F110 command, but it is absent.

### Expected results

Nothing appened, the flow remains always the same as default

### Add your .zip and screenshots here ⬇️

[EN4_Base sotto.zip](https://github.com/Ultimaker/Cura/files/14537462/EN4_Base.sotto.zip)
this is the file after i changed the inital flow rate to 110%. As you can see, the flow is always 94%, as default

GregValiant, 2024-03-08:
Thanks for the report.
When you set any "Flow Rate" in Cura, the amount is calculated into the "E" value in the Gcode lines.
M221 is a "global" command that would directly affect ALL the E values seen by the printer.
For example:  If you set the "Outer Wall Flow" to 115% in Cura, only the Outer Wall Flow is affected whereas M221 S115 would affect Outer, Inner, Skin, Skirt, Support, etc..  That would not be the user's intent.  In essence it would be necessary to have an M221 line at every change from one feature type to another.  That would be unwieldly at best.

If you use a post processor like "Change At Z" then yes, M221 is used as it is a global adjustment.

Your gcode file starts with "_M221 S94 ;Set the print traffic percentage_"
That will tell the printer to re-calculate all E values @ 94% flow.  That means every extrusion in the entire print will be 94% of whatever the Cura calculated flow is.  That will be true from your purge lines in the startup gcode through "G1 E-8 X5 Y5 Z3 F3000 ;Retract" in the ending gcode.  The entire file will be at 94% flow.  Every extrusion, and every retraction, and every prime will be at 94%.  If that works on your printer then you need to consider re-calibrating the E-steps.  

In the Settings at the end of the gcode I see "material_flow_layer_0 = 110" which is your setting in Cura.
The end result is that:
Initial Layer Flow = 94% of 110% = 103.4%
All other flows = 94% of 100% = 94%

I'll remove the bug label and close this as it appears to be just a misunderstanding of the setting.  I hope you understand.

closed 2024-03-08 by GregValiant

Paolinter, 2024-03-08:
Ok, thanks a lot for the explanation.So, how can i solve this problem? How can i set the flow of the first layer without changing the Gcode manually?

Yahoo Mail: cerca, organizza, prendi il controllo della tua casella di posta

GregValiant, 2024-03-08:
Change the line in your startup to M221 S100.  If you happen to tune a print while it's running, and you immediately print something else, it will ensure that the flow rate is set back to 100% for the new print. 
Set the Initial Layer Flow in Cura.  That's all you need to do.  The entire first layer will go down at that flow rate.

For reference - I always run the initial layer flow at 105% because I level with a piece of paper and I'm not always consistent.  The extra 5% flow really cuts down on the number of prints I have to abort because I got the Z gap wrong.