Ultimaker/Cura #18551: Silcer algorithm weak point in certain circuimstances

Zlutz, 2024-03-08:
### Cura Version

5.6.0 (same behaviour from 3.3.0 to 5.6.0, no issue with 2.5.0)

### Operating System

Windows 11

### Printer

Any

### Reproduction steps

Slice a part where:
1. Smaller piece (y) is on top of the bigger piece (x).
2. Use total wall thickness of (x-y)/2 - or more.

### Actual results

Top piece is printed directly on the infill of the bottom piece - there is no skin under the top piece, the walls of the top piece is not even touching the walls of the bottom piece!
I made a test file: 20mm cube on top of a 30mm cube and 5mm walls (10 x 0.5mm)
I attached a slicer screenshot - the infill "between" the cubes is actually visible!
Ironically - if I used half of the wall thickness (Which should be a much weaker part), there would be skin generated under the top piece and it would result in a much stronger part!

### Expected results

There should be skin under the walls of the top part - see photos attached

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2024-03-08:
Thanks for the report.
Without a project file I can't tell how you have it set up or what you are trying to do.
You have two separate models and one is resting on top of the other or is it a single model with a step?

This is a single model with the Skin Expand Distance set to 5mm.
[image]

If it is two separate models then you should get a full skin across the lower block.

I don't see a bug here and I'm inclined to close this as it appears to be a setting issue.  Please provide a project file.

Zlutz, 2024-03-19:
Sorry for not responding for so long, I had some medical issues...

Yes, it is a single piece, the cubes are a simple synthetic test I made so you can more easily understand what's happening.
If I want a very strong part, I'll put more walls and more top/bottom. This doesn't work if dimension changes in the part change at exactly the wall thickness point because then I don't have a wall, nor do I have Top/Bottom at the point.
I can't attach the test project file, it says "we don't support that file type".

You can easily reproduce it though, your example only has 3 walls so you don't see it - the problem happens when you use exactly (or more walls) so the outermost wall on the top piece is touching the innermost wall on the bottom piece, Skin Expand Distance doesn't work when there's no skin, and I consider "not generating skin under the walls" a bug.

I attached the STL file instead, you just have to set all the line widths to 0.5mm, and 10 wall line count.
It's also funny when you put 9 walls with "Alternate Extra Wall", 10 top skins, and skin expand to 10mm, then you get alternating layer of full skin, and then a layer with no skin, then full skin, then no skin - ten times.

I noticed the problem with some of my actual models, and it's especially bad when I need stronger smaller parts with curvy surfaces; everything works perfectly until the top piece of the part touches the curve of the bottom part - this "top" support disappears and the part is weaker as a result (and I'm adding walls to make it stronger because I need it stronger).

Image of the actual part, I think there should always be "yellow under green":
[image]

I mean, I can work around this issue by adjusting the number of walls, extra infill walls and similar. I fixed this "small part" by setting skin expansion to 100mm, but I don't really want to think about this every single time, especially with the more complicated parts.

p.s. I started "printing faster" by using only 2 outer walls and having thicker, faster extra infill walls. When I do it like that, I avoid most of this issue because then the top walls get printed on top of the infill walls and I get a stronger part.
I use "inside to outside" wall ordering (so I get a nicer outer surface and I can use thinner-than-nozzle outer wall line).
BUT, I would like to have "outside to inside" wall ordering for the infill (so excess material can get pushed inwards) because now it prints walls in order 2, 1 (outer), 5, 4, 3 (infill) - if I'm even slightly over-extruding, there's no room to shove the wall 3 between already printed walls 1, 2, 4 and 5 - and I get a skip or slip, printer flow control detects it, it goes cleaning, etc...

GregValiant, 2024-03-19:
I see it now.  As you say, it happens when Wall Count is just right when compared to the width of the shelf to the upper feature.
With this particular model - if the Wall Count is anything other than 10 the problem goes away.
There are probably other settings you could adjust to make a bad situation as well.

Zlutz, 2024-03-20:
That's true, but even 11 walls would create only 1 wall overlap an no skin under the rest of the walls - and there's still no top skin below! That's the problem, the whole part is 5.5mm thick, and "the weakest point" is only 0.5mm thick...
...I only made this "synthetic test example" to make the problem much clearer; in all probability, nobody ain't gonna do 10 walls anyways.

Let's talk something more realistic:
I need a 20mm axle with a narrowing for a 17mm bearing at the edge. That's a 1.5mm radius difference.
If you only use 2 x 0.5mm walls, the whole part is not be strong enough because 1mm is not that much for a part like this.
Then you up the wall count to 3 walls and you get this problem, making the part MUCH weaker than a 2 wall version.
Then you use 4 walls and that gives you 2mm of walls everywhere and 0.5mm overlap - STILL weaker than 2 wall version.
Then you use 5 walls, you get 2.5mm of walls and 1mm of overlap, and that's the same strength as the two wall part...
Only at 6 walls you get a part stronger than a 2 wall version, but only 50% stronger, so it's 300% more walls for 50% more strength.

This is the weakest point of the part, and "The chain is only as strong as it's weakest link"!

That's why I consider this a bug, it doesn't make sense.

Say I have a 20mm axle with 18mm narrowing, and a 14mm narrowing on top of that
2 walls - problem on 20-18 part
4 walls - problem on 18-14 part
6 walls - as strong as 2 walls... it makes no sense and what if I have a part even more complicated than that, there's no combination of settings I can do to make it strong everywhere, and I don't want to do the 100% infill...