Ultimaker/Cura #18559: Ultimaker Original Plus | gcode error prone, hotend stops heating after the 1st layer

bitSchleuder, 2024-03-09:
### Cura Version

5.6.0

### Operating System

MacOS Sonoma

### Printer

Ultimaker Original Plus

### Reproduction steps

1. Load 3D model from 3mf file
2. placed the model upside down
3. Setup of ABS filament (hotend 245° C, bed temp 80° C)
4. Start slicing
5. Saved to SD card
6. Put SD card into UltiController
7. Select gcode file
8. Start printing

### Actual results

Had to stop printing because the hotend temperature felled below 210° C. Setting the temperature manually won't work.
Temperatur goal is displayed, but hotend won't heat.

### Expected results

All ist running as expected, so the temperature will be held at 245° C, how it was set. Print will finish.

### Add your .zip and screenshots here ⬇️

[UMO_Goblin Halterung - Kein Senkkopf.gcode.zip](https://github.com/Ultimaker/Cura/files/14547704/UMO_Goblin.Halterung.-.Kein.Senkkopf.gcode.zip)

GregValiant, 2024-03-10:
Thanks for the report.
I'm not an "Ultimaker" guy but looking at the Gcode there is the Heat Up command in the StartUp section of the gcode and it properly sets the temperature to 245.
Cura adds another heating command at the end of the first layer (gcode LAYER:0).

In your gcode line 4690 is M104 S245 which is correct.  The line is there in case the "Initial Layer Temperature" is different than the "Print Temperature" and adjusts the temperature for the rest of the print.  Since both settings were the same in Cura, the print simply continues at the same temperature.

So the gcode is correct and it isn't apparent why the temperature would start to drop on the hot end.  Have you checked with UltiMaker support?  You might also ask for advice on the [UM Forum](https://commu nity.ultimaker.com/discover/).

I'll leave this open for now but the gcode looks exactly correct and consequently I don't see a Cura bug here so I will remove the bug label.  Let me know if you feel that is premature.
Your comment that the _"Temperature goal is displayed"_ would indicate something going on with the hardware since if the LCD is showing that the set point is 245 then the hot end should be 245.

GregValiant, 2024-03-13:
I'll go ahead and close this.  If you feel it should remain open let me know.

closed 2024-03-13 by GregValiant