Ultimaker/Cura #18561: The issue of running the source code for the Cura software.

Boyzkill, 2024-03-10:
### Is your feature request related to a problem?

The issue is that it has been impossible to successfully run the source code in accordance with the instructions provided on the GitHub website (https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source).

### Describe the solution you'd like

I would appreciate receiving a video tutorial on how to run the source code or an up-to-date working instruction on the process of running the source code for the Cura software.

### Describe alternatives you've considered

I attempted to run the application on two different operating systems, Windows and Ubuntu, as well as using different versions of Python. However, I encountered errors in all attempts.

### Affected users and/or printers

This information will be useful for all users

### Additional information & file uploads

The errors are presented in the word document
[erorr.docx](https://github.com/Ultimaker/Cura/files/14549692/erorr.docx)

ixd704, 2024-04-01:
Here is a working build recipe on Elementary OS (Ubuntu based) that I got to compile and run last night.  When you run it the first time, it will take a very long time to load as there are many hundreds of warning and error pages written to the terminal.  It will eventually load and run.  I have been successful in slicing a number of projects with this build.

You will note that I had to update to a newer version of cmake mid build which was not in the Ubuntu Jammy repository, so built and installed from source.  I had to adjust my PATH to include conan as it installed in a new directory.

My history is as follows (cleaned up to only include the required packages I needed to install):

 1887  pip install conan==1.60.2
 1888  conan config install https://github.com/ultimaker/conan-config.git
 1889  ls /home/idouglas/.local/bin
 1893  export PATH=/home/idouglas/.local/bin:/home/idouglas/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
 1894  conan config install https://github.com/ultimaker/conan-config.git
 1895  conan profile new default --detect --force
 1896  conan profile update settings.compiler.libcxx=libstdc++11 default
 1897  cd ~/Development/
 1898  git clone https://github.com/Ultimaker/Cura.git 
 1899  cd Cura/
 1901  conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
 1903  sudo apt install libx11-xcb1
 1904  sudo apt install libx11-xcb-dev
 1907  sudo apt install libfontenc-dev
 1909  sudo apt install libxaw7-dev
 1911  sudo apt install pkg-config libx11-dev libxkbfile-dev libsecret-1-dev
 1913  sudo apt install libxmuu-dev
 1915  sudo apt install libxcb-xkb-dev
 1918  sudo apt install libxcb-icccm4-dev
 1937  pip3 install sip
 1945  sudo apt install libxcb-*-dev
 1948  conan config install https://github.com/ultimaker/conan-config.git
 1949  conan profile new default --detect --force
 1950  conan profile update settings.compiler.libcxx=libstdc++11 default
 1951  conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
 1957  tar -xvzf cmake-3.28.4.tar.gz 
 1958  cd cmake-3.28.4/
 1960  mkdir build
 1961  cd build/
 1962  cmake ..
 1964  make -j8
 1965  sudo make install
 1966  sudo apt remove cmake
 1967  cmake --version
 1968  /usr/local/bin/cmake --version
 1969  cd ../..
 1970  cd Cura/
 1971  history 
 1972  conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
 1973  ~/Development/Cura$ source venv/bin/activate
 1974  ~/Development/Cura$ python cura_app.py

Boyzkill, 2024-04-04:
> В случае, если кто-то запрашивает загрузку выходных данных, сгенерированных в то время, как загрузка первого запуска занимает вечность, вот оно: Слишком длинный комментарий, поэтому файл прилагается, [Terminal_ouput_dump.txt](https://github.com/Ultimaker/Cura/files/14823532/Terminal_ouput_dump.txt)

Thank you for the information provided. I wanted to know if you still have a virtual machine, if so, could you send it?

ixd704, 2024-04-04:
I am not using a virtual machine.  I run Linux as my base OS.

ixd704, 2024-04-04:
You can get it from the Cura site.  Just download the Linux app image file.  Change it to be executable with:
chmod +x <filename you downloaded>
then
appimagefilename --appimage-help
Will list what you can do with the image file

appimagefilename --appimage-extract
It will extract the app image contents.

The first time you run it, let it run for a while as it make look like it is frozen, but it is probably not as there is a load of items it is setting up and configuring.

Boyzkill, 2024-04-04:
Thank you very much

ixd704, 2024-04-04:
What I did was compile it on my machine.  To do that, you will have to install all of the packages I did and setup the virtual environment for it to be able to run.

The appimage route is far easier.  My build is 5.7, so the same as the app image.

ixd704, 2024-04-04:
Here is a link to app image info....
https://superuser.com/questions/1301583/how-can-i-extract-files-from-an-appimage

Boyzkill, 2024-04-04:
Thank you for your help