Ultimaker/Cura #18570: Repeat printing on the same layer

Yyzinfo, 2024-03-11:
### Cura Version

5.6.0

### Operating System

windows11

### Printer

diy

### Reproduction steps

Repeat printing on the same layer

https://github.com/Ultimaker/Cura/assets/163029026/4976b498-bf19-4cd4-9281-f5960d5b4f3b

### Actual results

Repeat printing on the same layer

https://github.com/Ultimaker/Cura/assets/163029026/851d6a0c-9716-4c98-a7c5-6e27f5cee231

### Expected results

Repeat printing on the same layer

### Add your .zip and screenshots here ⬇️

Repeat printing on the same layer

GregValiant, 2024-03-11:
Thanks for the report.
You will need to provide a project file or at least the gcode file.  Without them there can be no investigation.

Yyzinfo, 2024-03-12:
> 感谢您的报告。您需要提供一个项目文件或至少一个 gcode 文件。没有他们，就没有调查。

LAYER:86
[bug.zip](https://github.com/Ultimaker/Cura/files/14567624/bug.zip)

GregValiant, 2024-03-12:
Because you have "Arc Welder" enabled Cura will not display the gcode correctly (it doesn't understand G2 or G3 moves).
Please post the project file ("File | Save Project").
I do see the "double interface" in the preview.  I haven't seen that before so to figure out why it's there I will need to know the settings you used.  They will be in a project file.

Yyzinfo, 2024-03-12:
> 由于您启用了“电弧焊机”，因此 Cura 将无法正确显示 gcode（它无法理解 G2 或 G3 移动）。请发布项目文件（“文件 |保存项目“）。我确实在预览中看到了“双界面”。我以前没有看到过，所以要弄清楚为什么它在那里，我需要知道你使用的设置。它们将位于项目文件中。
> 
> 从 gcode 中的第 1371 行开始，我看到了这一点。 G92 E0 G1 F2400 E-11 G92 E0 G1 F2400 E16 G1 F2400 E0 G1 F2400 E16 G1 F2400 E0 G1 F2400 E16 G1 F2400 E0
> 
> G1 F2400 E16 G1 F2400 E0 G1 F2400 E16 G1 F2400 E0 G1 F2400 E16 G1 F2400 E0
> 
> G1 F2400 E16 G1 F2400 E0 G1 F2400 E16 G1 F2400 E0 G1 F2400 E16 G1 F2400 E0
> 
> 我不明白这样做的必要性。

Just let the consumables cool down in a specific place。
3mf file 
[CFFFP_bug-3mf.zip](https://github.com/Ultimaker/Cura/files/14574245/CFFFP_bug-3mf.zip)
[image]

GregValiant, 2024-03-12:
I can reproduce this with the project file.  I switched to a couple of different dual-extruder printers and the problem went away.  When I switched to an Ultimaker S5 then the problem showed up again.

@HellAholic you may want to look at this one.

HellAholic, 2024-03-13:
Hey @GregValiant , thanks for the heads up. We're working on several tickets related to fractional z distance and support interface, I've added this one to the list as well. Internal reference: CURA-11735

HellAholic, 2024-03-26:
Hey @GregValiant & @Yyzinfo,
We did some fixes and suddenly a @ThomasRahm appeared with a PR for tree support that also fixes this issue.
They will be included in the stable release for 5.7
Thanks for the report and the heads up ;)

Cheers!

closed 2024-03-26 by HellAholic