Ultimaker/Cura #18584: Head

olavoabrantes, 2024-03-12:
### Cura Version

5.6.0

### Operating System

Mac OS Sonoma

### Printer

Ultimaker 2+ Connect

### Name abnormal settings

The model doesn't slice

### Describe model location

The model doesn't slice even if rotated

### Describe your model

I haven't managed to slice it

### Add your .zip here ⬇️

[Head.3mf.zip](https://github.com/Ultimaker/Cura/files/14579064/Head.3mf.zip)
[Head.3mf.zip](https://github.com/Ultimaker/Cura/files/14579073/Head.3mf.zip)

GregValiant, 2024-03-13:
Thanks for the report.
The model has errors in it.  After repairing the model 5.6.0 still would not slice it.  The 5.7alpha release did slice it so hopefully this will be fixed when 5.7a is officially released.

I'll mark this as a duplicate report.

olavoabrantes, 2024-03-13:
Hello, thank you for letting me know.

All the best,
Olavo