Ultimaker/Cura #18590: Saved project does not maintain item positions

keith20mm, 2024-03-13:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Ender-3

### Reproduction steps

Place two or more items onto build plate.
Situate one item onto the build plate.
Situate other item(s) above (z-axis) on the top of the other, and requires uncheck of Drop Down Model.
Save the project, and exit Cura.

### Actual results

Restart Cura and File | Open File the saved project .3mf

The items all are now on the build plate, instead of in their saved z-location(s)

### Expected results

Restart Cura and File | Open File the saved project .3mf

The items should be positioned in their saved z-location(s)

### Add your .zip and screenshots here ⬇️

[Uploading FawnOnRestForTest.zip…]()
[image]
[image]

GregValiant, 2024-03-13:
Thanks for the report.
This behavior has been reported previously (#18419) and it has been mentioned in other reports.  Hopefully this will be addressed in 5.7.

I'll mark this as a duplicate report.

modrows1, 2024-03-23:
> Thanks for the report. This behavior has been reported previously (#18419) and it has been mentioned in other reports. Hopefully this will be addressed in 5.7.
> 
> I'll mark this as a duplicate report.

Still occuring in 5.7.0 beta - also shifting support blocks in all directions not just Z