Ultimaker/Cura #18595: Cura 5.6.0 -  Spiralize feature creates random star burst at top of model

gr8design44, 2024-03-13:
### Cura Version

5.6

### Operating System

MacOS 14.4

### Printer

Neptune 4 Max

### Reproduction steps

1. Load attached Sound Panel model.
2. Choose "Spiralize Outer Contour."
3. "Slice" then "Preview"

### Actual results

Slicer adds a starburst pattern from the corner the Spiralize feature finishes at. This does not happen in Cura 4.8 (Elegoo).

### Expected results

The model should not have a starburst pattern on the most top layer.

### Add your .zip and screenshots here ⬇️

STL and screenshots in zip folder

GregValiant, 2024-03-13:
Thanks for the report.
The starburst exists on the top two layers because the model comes to a knife edge.  The last two slices can't fit two moves (out and back) and so the nozzle must "travel" to the next "island".  Since you are in spiralize, none of those are travel moves but rather extrusions.  That starburt will be in the print.
You will need to play a bit and either subtract a little (0.6mm??) from the top of the model so there is a flat area on top (and no knife edges), or you can add a cutting mesh in Cura and set everything to 0 so the top problem layers don't print.

This is with a cutting mesh and it's 22.4 off the build plate.  This corner is the problem.  The next layer would be too narrow and the path will develop an "end" and so the starburst will appear.
[image]

This is the "new" top layer and there is no starburst.
[image]

I think this is a case of model geometry that isn't suitable for spiralize (once slicing gets above the 22.4mm height).  There are things you can do to fix this problem so I'm going to remove the bug label.  Let me know if you feel that is premature.  I'll leave this open so you can respond.

gr8design44, 2024-03-13:
Hi Greg.Wow! Thank you for the detailed explanation. I really appreciate it and it does make sense. Why do you suppose it doesn’t happen in Cura 4.8? I was able to print them at .28 with spiralize on. I followed the designers setting from Thingiverse.Thanks.Rich

image.png (view on web)
image.png (view on web)

GregValiant, 2024-03-13:
Spiralize was reworked for (I think) 5.3.0.  It now utilizes "Remove all Holes" to make more models amenable to spiralize.
There is a serious side effect to that in regards to trying to put holes in things like a lamp bottom.  This might be another side effect but this is the way it works now.
Knife edges / Feather edges are never a good idea in FDM.  Printing screw threads can really be a problem if they don't have flats at the root and at the apex.

gr8design44, 2024-03-14:
Ok, got it! Thank you so much for your time spent explaining to a newbie. Have a gr8 evening!Rich

GregValiant, 2024-03-14:
I'll go ahead and close this report.

closed 2024-03-14 by GregValiant