Ultimaker/Cura #18600: dual head priming with the wrong head

illusionmanager, 2024-03-13:
### Cura Version

5.6.0

### Operating System

windows 10

### Printer

snapmaker with dual head

### Reproduction steps

created a thin 3-layer box of blue filament with on top small white elements (making a ruler). The blue part was set to be done with a 0.4 nozzle, the white part with a 0.2 nozzle.
So three layers in blue, followed by three layers in white on top of the blue.

### Actual results

what it did after finishing the ironing of the blue box, it went back to the prime tower, printed the tower element in blue, then switched to the white nozzle and only printed a single dot on the prime tower. 

### Expected results

it should have primed the white instead of the blue when doing the white part.

### Add your .zip and screenshots here ⬇️

the screenshot shows the very 4th layer which should have printed in only white, instead it did the prime tower in blue and basically skipped the priming of the white. The different nozzle sizes might trigger this, I don't know. it just didn't prime the white nozzle.
[test-ruler.zip](https://github.com/Ultimaker/Cura/files/14594920/test-ruler.zip)

GregValiant, 2024-03-14:
Thanks for the report.
This looks to be a settings issue.  The file you provided is just the Model file.  We need a Project File to troubleshoot.
Use the "File | Save Project" command to create a project file, zip it, and post the zip folder.

I don't have a Snapmaker definition file so I used a different dual extruder printer.  You can see that the purge on layer 4 is the same for both extruders.  The 5 narrow lines are the 0.2 nozzle of extruder 2 and the wide lines are the 0.4 nozzle of Extruder 1.  The volumes would appear to be equal.
I cannot duplicate the problem you are having.  I'll leave the bug label on this for a bit but post the project file.
[image]

illusionmanager, 2024-03-14:
Thanks for taking the time to look into this. here is the snapmaker definition that I used with cura:  https://github.com/Snapmaker/SHapmakerCuraPlugin/releases. The problem is more easily observed when using the material color to visualize the preview (instead of line type). It might be a problem with the snapmaker plugin, or with cura. If it is a user setting, I don't know what could have caused this.

closed 2024-03-14 by GregValiant

illusionmanager, 2024-03-14:
just found the problem... As I only needed white on top of the blue, I didn't need a large prime tower, so I set its size to 10mm, enough to get the white flowing. However cura decided to prime blue first (totally unnecessary) followed by the white. Because my tower was small there wasn't room left for the white priming. Making the tower wider solves the problem. It still primes the blue first, but then there is enough priming of the white part. So the problem is solved, although cura could be a bit improved by not priming the blue, when it needs to use the white.

illusionmanager, 2024-03-14:
yes please close this as a bug. perhaps leave it as a request for improvement

GregValiant, 2024-03-14:
The prime tower always has to grow with the print.
The next person who posts will want it switched to the way it currently is.  The software has to consider all situations and as a consequence 1/2 the people are unhappy 1/2 the time.

illusionmanager, 2024-03-14:
its not a big issue at all, but I fail to see why you still need to print blue on the tower when it is the last blue to be printed. (in the layers above it doesn't print the blue, not even on the tower). It would be only a tiny optimization, saving a bit of print time and a bit of filament. The model is now fixed, by creating a union of all the elements.