Ultimaker/Cura #18611: Top & Bottom layers on surface mode

NTD3D, 2024-03-14:
### Is your feature request related to a problem?

top bottom layers on surface mode

### Describe the solution you'd like

I'm printing a wing for a plane (LW-PLA) using surface mode, this works great, but i cant apply a top and bottom layer. when i use standard mode with a 0% infill and single wall i get a very ugly finish. i have added a picture for reference.

if there is a setting already for can you let ne know what it is, if not is it possible to have it added (plane mode)

every slice currently is one continous motion making the stop start position the same on ever slice.
[image]

### Describe alternatives you've considered

using normal mode, but i cant get it to give the result i want

### Affected users and/or printers

it will help anyone who 3D prints planes, as none of the free slicing softwares offer this feature.

GregValiant, 2024-03-15:
I used to build planes.  In FDM they present some unique challenges.

For both "Surface Mode" and "Spiralize Outer Contour" it's a function of the angle to the build plate.  As the angle approaches 0° the distance between the layer "steps" approaches infinity.  Since the "Step Height" is the Layer Height, that setting also comes into play.  This is much the same as determining the maximum "Support Overhang Angle".

This is a horizontal round protrusion in "Surface Mode" and 0.30 layer height.  The distance between the extrusions varies because the math says they must.
[image]

At .05 layer height the problem is better, but it can't go away.
[image]

With a Support Blocker configured to "Modify Settings for Overlaps" and "Cutting Mesh" and with the setting "Surface Mode" set to "normal" for the blocker, you might be able to at least mitigate the problem.
[image]

NTD3D, 2024-03-15:
i want to add the top layer but when i do this it adds top layer to all of the angled layers which causes retracts and rapid moves which gives stringing, if you take a look and my image you'll see one retract point, this stays the same the whole way up. 

thank you for your response