Ultimaker/Cura #18612: Unexpected error when slicing

Aurarum, 2024-03-14:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

reprap based Aurarum 3d printer

### Reproduction steps

1. loaded a file
2. try to slice

### Actual results

Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.

### Expected results

sliced file

### Add your .zip and screenshots here ⬇️

[VEGA600_Basketball_ver_0.5.zip](https://github.com/Ultimaker/Cura/files/14608042/VEGA600_Basketball_ver_0.5.zip)

GregValiant, 2024-03-15:
Thanks for the report.
This is the 4th report regarding what appears to be the same model of a basketball.
Please see #16389 , #18449 , #18510.
In the 18449 report a member of the Cura team says that the next release will contain a fix for this.
One of the basketball models had over 91,000 errors in it.  Not slicing that one can't really be called a bug because the model is so bad.

This model is "not watertight" and so it really needs repair.  I don't see all the errors of that first model, but it does have some errors.

Aurarum, 2024-03-15:
You are correct - fixing the model fixes the issue.

closed 2024-03-15 by Aurarum