Ultimaker/Cura #18621: Don't have access to support settings....

demetan71, 2024-03-15:
### Cura Version

5.6

### Operating System

windows 11

### Printer

Anycubic Chiron wiht skr board

### Reproduction steps

I want to change support settings because I'm unable to remove it without damage the part.....  I'm always have to change for another slicer for support.  Try many version of Cura, the 5.6, today I switched to 5.4, same thing, no access to support settings
[image]

### Actual results

I want to change support settings because I'm unable to remove it without damage the part.....  I'm always have to change for another slicer for support.  Try many version of Cura, the 5.6, today I switched to 5.4, same thing, no access to support settings
[image]

### Expected results

I want to change support settings because I'm unable to remove it without damage the part.....  I'm always have to change for another slicer for support.  Try many version of Cura, the 5.6, today I switched to 5.4, same thing, no access to support settings
[image]

### Add your .zip and screenshots here ⬇️

[image]

demetan71, 2024-03-15:
Sorry, it work with version 5.4, but I don't know how to delete this tread!!!

HellAholic, 2024-03-16:
Hey @demetan71 
When support is disabled, all the settings regarding the support are hidden/not displayed. Additionally, you can change the view to show you all of the settings instead of the basic view from the burger menu on top right of the settings view.

[image]
[image]
[image]

Glad you found a workaround, will close this issue.

closed 2024-03-16 by HellAholic