Ultimaker/Cura #18623: [0.5.6] Slicing with `Experimental`/`Alternate Wall Directions` does not respect seam location settings

Sophist-UK, 2024-03-15:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Dagoma DE200

### Reproduction steps

1. Print cube with 2 walls and 0 top surfaces
2. Set Experimental/Alternate Wall Directions and e.g. Walls/...
    * Z-Seam Alignment to User Specified
    * Z-Seam Position to Back Right
    * Seam Corner Preference to Expose Seam
 3. Slice

### Actual results

Should be no travels from one corner to another but there are.

Play with the seam values and see inner and outer walls having different seam points etc.

### Expected results

No travels from one corner to another.

### Add your .zip and screenshots here ⬇️

[image]

Hal9999, 2024-04-07:
I can confirm this issue.

[image]
[image]

OsmosisD, 2024-06-17:
Can also confirm. Unexpected direction changes partway along a print totally ruins the results.
[image]
Z-Seam location set to center of the print plate in this example.
[image]
Print results (one canceled due to head crash).