Ultimaker/Cura #18627: Mounting bracket print fail to slice

sstarkny, 2024-03-16:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 3

### Name abnormal settings

tree supports, brim enabled

### Describe model location

lip of the bracket faces up

### Describe your model

Breadboard mounting bracket

### Add your .zip here ⬇️

[breadboard_mount_rev_2.zip](https://github.com/Ultimaker/Cura/files/14621826/breadboard_mount_rev_2.zip)

GregValiant, 2024-03-16:
Thanks for the report.
There have been numerous reports regarding Cura having trouble on Win11 systems.
I'll mark this as a duplicate as I didn't have any trouble slicing the model on my Win10 system.
The Cura team is aware of this bug and hopefully it will be repaired in 5.7.

MariMakes, 2024-03-18:
Hi @sstarkny 

Thanks for your report 👍
Several slicing crashes have been resolved with the 5.7 Beta release.

I was able to slice your file in that version of Cura. 
[image]

You can download the Cura version here: [https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta.1)

Can you let us know if that version of Cura resolves your issue? 

Thanks again, and please let us know if you run into any other issues 💪

closed 2024-04-02 by github-actions[bot]