Ultimaker/Cura #18630: Slicing Fehler

supermarioiv, 2024-03-16:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Sovol SV02

### Name abnormal settings

keine unnormalen Einstellungen

### Describe model location

Bahnübergang LGB Spur G

### Describe your model

Es ist ein Ganz normales Modell, ohne Besonderheiten.

### Add your .zip here ⬇️

[Gb-Bueber s2.zip](https://github.com/Ultimaker/Cura/files/14622957/Gb-Bueber.s2.zip)

GregValiant, 2024-03-16:
Thanks for the report.
The model is not watertight and really needs to be repaired.  You can use MS 3D Builder which has a good repair utility.
Try moving or rotating the model on the Cura build plate.  Sometime changing the Initial Layer Height by a little bit can free up slicing as well.
I didn't have any trouble slicing the file.  This is in 5.6.0 and on a Win10 computer.
[image]

MariMakes, 2024-03-18:
Hi @supermarioiv, 

Several slicing crashes have been resolved with the 5.7 Beta release.
You can download the Cura version here: [https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.7.0-beta.1)

Can you let us know if that version of Cura resolves your issue? 
Thank you 🙏

closed 2024-04-02 by github-actions[bot]