Ultimaker/Cura #18632: cant export printer profiles

91crrider, 2024-03-16:
### Cura Version

5.5.0

### Operating System

windows 10

### Printer

Ender3/ender3 v2

### Reproduction steps

just try and export it

### Actual results

failed wont actually export

### Expected results

saved to folder i selected 

### Add your .zip and screenshots here ⬇️

wont save

HellAholic, 2024-03-16:
Hey @91crrider ,
I tried and it exported.
[image]
Please include the steps your taking otherwise it is impossible to help you with your issue.

Potential sources for your issue:
- You do not have space left on the drive you're trying to export to
- You do not have permission to write to the location you're trying to save to
- You have an antivirus or another software running that is preventing you from writing to that location

91crrider, 2024-03-16:
Sorry for the inconvenience I did figure out this issue, and it was not on
cura's side but I really appreciate the rapid response and for any future
issues I will include the screen shot thank you again

GregValiant, 2024-03-16:
I'll go ahead and close this as it appears to be taken care of.

closed 2024-03-16 by GregValiant