Ultimaker/Cura #18635: Pointless support

pcristea85, 2024-03-16:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 5 Plus

### Reproduction steps

Print a block detached from the build plate with a horizontal hole in it. Select support only touching base plate.

### Actual results

Even though I don't want to have a support inside the hole in the middle, the support structure still reaches up to that hole where support would have been generated if I did not select "only touching baseplate".

### Expected results

Like in Cura 5.1.0, I expect the support structure to only reach up to generated supports. Something changed between these versions as the same settings produce different results.

### Add your .zip and screenshots here ⬇️

[image]

Asterchades, 2024-03-17:
Try toggling `Enable Towers`. It's currently not set to be visible in your settings, so you will need to either use the search box or set it to be visible so you can adjust it.

While there have been adjustments made to Towers during the Cura 5 cycle they're still very hit-and-miss in my experience, and this looks exactly like one of the misses.

HellAholic, 2024-03-17:
You can add a support blocker to sections you don't want to be supported.
Support placement only affects whether the support is placed on the model as well or only on the build plate.

[image]
[image]
[image]

pcristea85, 2024-03-17:
> Try toggling `Enable Towers`. It's currently not set to be visible in your settings, so you will need to either use the search box or set it to be visible so you can adjust it.
>
> While there have [...]

Thanks. That did it.

@HellAholic Thanks for the detailed response with screenshots. I am aware how to use support blockers but I don't think using them in this context is good UX because it's very time consuming and in some instances not even a solution. Like I said it used to work out of the box. Hopefully it will be fixed and restored back to normal. Or perhaps make the towers option visible by default, because if I saw it, I would have figured it out. Other users will have similar experiences.

Asterchades, 2024-03-18:
Not sure if I should close this one or not. On the one hand the immediate issue has been resolved, but on the other it shows there's still situations where Towers produce undesirable - and unexpected - results.

I'll defer judgement on that to someone more closely involved with the project.