Ultimaker/Cura #18637: Cura missing build plate once opened

basilverthorn, 2024-03-16:
### Cura Version

5.7.0 Alpha & Beta

### Operating System

Arch Linux

### Printer

Ender 3v2 and XMAX 

### Reproduction steps

Installed latest version 5.6 after updating to Plasma 6.X
Installed next few pre releases to see if the Alpha or Beta would address the issue

### Actual results

In 5.6 Cura opens to 100% See thru Window
In 5.7.X is looks better but no build plates appear for any of my printers. 
Open a model it does not show either

### Expected results

A build plate would be visible

### Add your .zip and screenshots here ⬇️

[image]
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/14625453/cura.log)

Arch linux, nvidia driver DKMS, same issue under Wayland, Hyprland, Qtile.  Older skined version of Cura by Qidi, works just fine.

basilverthorn, 2024-03-16:
I see in the logs it is complaining about OpenGl. Not sure if this helps but posting anyway:

local/glew 2.2.0-6
The OpenGL Extension Wrangler Library
local/glib2 2.78.4-1
Low level core library
local/glibmm 2.66.6-1
C++ bindings for GLib
local/lib32-glew 2.2.0-4
A cross-platform C/C++ extension loading library (32 bit)
local/lib32-glib2 2.78.4-1
Low level core library - 32-bit
local/lib32-libglvnd 1.7.0-1
The GL Vendor-Neutral Dispatch library
local/lib32-mesa 1:24.0.3-1
An open-source implementation of the OpenGL specification (32-bit)
local/lib32-nvidia-utils 550.54.14-1
NVIDIA drivers utilities (32-bit)
local/libglvnd 1.7.0-1
The GL Vendor-Neutral Dispatch library
local/mesa 1:24.0.3-1
An open-source implementation of the OpenGL specification
local/nvidia-utils 550.54.14-5
NVIDIA drivers utilities

x glxinfo | grep "OpenGL version"
OpenGL version string: 4.6.0 NVIDIA 550.54.14

~
> glxinfo | awk '/OpenGL version/ {print $4}'
4.6.0

sausix, 2024-04-12:
Affects me too. White page in pure wayland mode as in screenshot. On forcing to xwayland heavy flickering and pure white moments until camera angle changed a bit.

Same config: Arch Linux, KDE Plasma, nvidia-dkms

nallath, 2024-04-15:
What I can find from the logs is that it completely fails to create an opengl context. I'm not quite sure what is causing it.

That being said, are you using the appimage or some other distro?

sausix, 2024-04-15:
5.7.0 AppImage is used as install source: https://aur.archlinux.org/packages/cura-bin
I've just tried the official 5.5.0 AppImage directly and Wayland and XWayland both are not usable.

With `xcb` it's flickering as in 5.7.0. Pure Wayland only the window border is being rendered.

basilverthorn, 2024-04-15:
I have tried both.  Both do the same, the only version that works for me is what ever version Qidi has customized in their appimage.  Of course I can’t use it with my Enders :-(.  Sent from my iPhone

sausix, 2024-04-29:
I found a workaround and posted here: https://github.com/Ultimaker/Cura/issues/18891#issuecomment-2083484534

MariMakes, 2024-05-01:
🙏  Thanks for the super helpful comment @sausix 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/18891