Ultimaker/Cura #18639: dual head nozzle switching at wrong position

illusionmanager, 2024-03-17:
### Cura Version

5.6.0

### Operating System

windows 10

### Printer

snapmaker with dual head

### Reproduction steps

printed something with my dual head

### Actual results

when it switch for the first time from the blue nozzle to the white nozzle, the nozzle was right above a printed blue part. At that position it started to heat up the white nozzle, oozing a bit of white material right on top of the blue part, before it started printing a brim in white.

### Expected results

it should have moved the nozzle out of the way, or perhaps to the start of the brim, but not stay above the last position of the already printed blue part.
Preferably it should move to an empty spot whenever it is waiting for a nozzle to heat up, which includes waiting for a nozzle to go from standby temperature to printing temperature after a nozzle switch.

### Add your .zip and screenshots here ⬇️

[ruler.zip](https://github.com/Ultimaker/Cura/files/14626821/ruler.zip)

GregValiant, 2024-03-17:
Thanks for the report.
It appears that the M109 line is a problem on this particular print.  M109 tells the printer to wait for the temperature set point to be reached.
Even when the temperature is already correct - M109 can cause a 10 second delay before the print resumes.  On this model it is a problem.  Cura uses M109 because on prints with short layer times the nozzle can't heat up in a timely fashion so a delay to wait for the temperature is necessary.

There are two workarounds for this.

1. In the Machine Settings add a move to the Extruder Start Gcodes for both extruders.  A move off the print like:
G1 F{speed_travel*60} X0 Y0 should work.
3. Use "Search and Replace" and search for M109 and replace it with M104.  You will need to go into the gcode to ensure that the M109 in the printer StartUp gcode is put back in place.  You don't want to start with a cold extrusion.

I'll leave the bug label on this but it may be something that SnapMaker should have addressed in their definition files.  The Cura team will take a look.

This is the standard gcode that is causing your problem:
T1
G92 E0
M105
**M109 S197**
M104 T0 S150
G1 F2100 E-1.5
;MESH:NONMESH
G0 F3600 X232.001 Y66.256 Z1.2

Altering the Extruder StartUp Gcode results in this:
T1
G92 E0
**G1 F3600 X0 Y0**
M105
**M109 S197**
M104 T0 S150
G1 F2100 E-1.5
;MESH:NONMESH
G0 F3600 X232.001 Y66.256 Z1.2

illusionmanager, 2024-03-17:
thanks for this very good reply. I'll forward it to the snapmaker support so they can change it.
I'm no expert, but just to make sure, your first fix only works if the rest of the code doesn't rely on the position the nozzle used to be, as you now moved it to 0,0. This is probably true in most cases, I just don't know if that is true in all cases.

GregValiant, 2024-03-17:
That is the problem with trying to move off to the side.  In this case there is a move to an extrusion start position so it isn't an issue.  That may not be true for every print but is probably true for 99.99% of them.
It can't be a "relative" move because there isn't any way to work the current location into a move.