Ultimaker/Cura #18641: Options cancel too quickly

benclewett, 2024-03-17:
### Cura Version

5.7.0-beta1

### Operating System

MacOS 14.2.1

### Printer

Creality v2 Neo

### Reproduction steps

Select an option, like Infill density.  Delete the current option.  Wait about a 0.1 of a second.  The value will be replace with 0 and the option committed.  It gives almost no chance in typing in the option I want.

### Actual results

As above

### Expected results

As above

### Add your .zip and screenshots here ⬇️

None

GregValiant, 2024-03-17:
Thanks for the report.
I know I can reproduce this one.  As soon as you get one character typed into a box you get bopped out and the rest of your typing ends up on the floor.

The Cura team will take a look.

MariMakes, 2024-03-18:
Hi @benclewett 

Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12949
We have a ticket on our backlog with the intent to improve this behavior.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12949

closed 2024-03-18 by MariMakes

benclewett, 2024-03-19:
Thanks.  It seems a shame that such a high-impact bug is left unfixed for
nearly 20 months.